uw-biorobotics/IKBT #5: 7dof Support

mdicicco, 2018-06-19:
Exciting project.  I was curious if 7dof robots (or, really anything greater than 6) are supported.

I built a robot in the ik_robots.py with a theta_7 variable and d_7 parameter which both caused errors in variables being not defined.  It looks like you have global variable lists in both ik_classes.py (line 40) and ikSolver.py (line 64).  Those duplicated globals are probably something worth addressing in the future.

In the meantime, I added my variables and the code was able to run, but then it failed with the following error:
`currently looking at:  th_2
basenode:  X2Y2 id and solver  FAIL
Traceback (most recent call last):
  File "ikSolver.py", line 358, in <module>
    final_groups = matching.matching_func(R.notation_collections, R.solution_nodes)
  File "/Users/mdicicco/IKBT/ikbtbasics/matching.py", line 128, in matching_func
    start_list = notation_d[max_len] # get lists with most variables
KeyError: -1
`
Is this maybe another error related to the code, or is the kind of error I would expect from trying to solve a redundant manipulator?

dianmuz, 2018-06-19:
Thanks for the interest!

Currently IKBT doesn't support 7-DOF automatically. To be exact, you need to "guess" one DOF (out of the >6 DOFs) for any algorithms, since any DOF number that's greater than 6 would have infinite number of solutions. 

In short it's not an error of the code. (My guess is if you set one of the DOF as known, and only put in 6 variables for IKBT. It's likely going to solve it.)

mdicicco, 2018-06-19:
Okay, thanks for the clarification.  Feel free to close this issue, then.

You might want to consider a feature in the future where you can specify this kind of free variable.  That's how I've seen IKFast and a few other hand coded IK solutions do it.  When you make the final IK call, you provide the transform for the end-effector along with specified values for all the free variables and then it solves for the other 6 non-free variables.

dianmuz, 2018-06-19:
Thanks for the pointer! Very helpful, I'll definitely look into it :) 

7 DOF support is actually one of things we're planning to build, along with 3+ parallel axes.  

(BTW, you can actually run IKFast?? We tried to do it, but it always asked us to downgrade several packages.  Because we have other things that depend on current versions of those packages, and the OpenRave is too gigantic for the virtual machine, we eventually gave up. I'm interested in learning how to do it without downgrading!)

mdicicco, 2018-06-19:
Closing this issue now.  Keep in mind that the globals problem I mention in my original comment is going to get in the way as you go to a larger number of joints.  You may want to define those globals in one file and do it programatically (just define a_n, d_n, th_n for all n up to the max_dof).

If you don't want to go full support for 6+n joints (where all n would be specified at IK call time) I think that just a special case for 7 would be very useful.  Lots of robots on the market now use 7.  I have a KUKA iiwa which is 7, and the Rethink robots are both 7, along with the PR2 arms.

I did get IKFast to work a long time ago, but I needed to get some email support from the developer to get it to run successfully.  If I recall, the error had something to do with symbolic solver having problems with a mismatch the number of decimal points used to define the constants.  My problems running IKFast were one of the reasons I was excited by your project.

closed 2018-06-19 by mdicicco