uw-biorobotics/IKBT #6: SyntaxError: Missing parentheses in call to 'print'. Did you mean print(int "")?

amin101, 2018-07-24:
running code `python ikSolver.py wrist` throw error:

` File "ikSolver.py", line 50
    print ""
           ^
SyntaxError: Missing parentheses in call to 'print'. Did you mean print(int "")?`

blake5634, 2018-07-24:
IKBT Requires Python 2.7.X 
(one of the biggest differences between 2.7.X and 3.X.X is 
2.7.X:    print "message"
vs.
3.X       print ("message")    (!)

Thanks for trying it out!

amin101, 2018-07-24:
Thanks for reply. after installing python 2.7.15 see this error:
`
          Running IK solution

             Working on Wrist

kinematics pickle: trying to open  fk_eqns/Wrist_pickle.p  in  C:\Users\lion\Desktop\IKBT-master
Did not find VALID stored pickle file:  fk_eqns/Wrist_pickle.p
Starting Forward Kinematics
Traceback (most recent call last):
  File "ikSolver.py", line 109, in <module>
    [M, R, unknowns] = kinematics_pickle(robot, dh, params, pvals, vv, unknowns, testing)
  File "C:\Users\lion\Desktop\IKBT-master\ikbtbasics\ik_classes.py", line 90, in kinematics_pickle
    m.forward_kinematics()
  File "C:\Users\lion\Desktop\IKBT-master\ikbtbasics\kin_cl.py", line 160, in forward_kinematics
    self.T_06 = sp.trigsimp(self.T_01 * self.T_12 * self.T_23 * self.T_34 * self.T_45 * self.T_56)
  File "C:\Python27\lib\site-packages\sympy\simplify\simplify.py", line 902, in trigsimp
    result = _trigsimp(expr, deep)
  File "C:\Python27\lib\site-packages\sympy\simplify\simplify.py", line 956, in _trigsimp
    if expr.is_Mul:
  File "C:\Python27\lib\site-packages\sympy\matrices\matrices.py", line 3008, in __getattr__
    raise AttributeError("Matrix has no attribute %s." % attr)
AttributeError: Matrix has no attribute is_Mul.`

amin101, 2018-07-25:
uninstall python 2.7.15 and install python 2.7.14, then install `sympy` package through CMD `pip install sympy` (not installer) solved the issue.

closed 2018-07-25 by amin101