uw-biorobotics/IKBT #7: Solution finder does not work

VidyaPuri, 2018-07-26:
Greetings,

i tried using your IKBT solver but no matter what robot example i use, i do not get any valid results.

i tried three different robots, with their DH parameters. One was from the book from our University with the following DH parameters, the robot described was supposed to be Puma-like robot.

`     if(name == 'PumaV2'):`
       ` dh = sp.Matrix([`
         `   [ sp.pi/2 ,    0 ,  d_1 ,      th_1  ],   #  Note: Puma is used for tests so mods to this table`
         `   [       0 ,  a_2 ,    0 ,      th_2  ],   #  may break ikbtleaves.updateL.TestSolver007`
        `   [-sp.pi/2 ,    0 ,    0 ,      th_3  ],   `
         `   [ sp.pi/2 ,    0 ,  d_4 ,      th_4  ],      `
          `  [-sp.pi/2 ,    0 ,    0 ,      th_5  ],`
           ` [       0 ,    0 ,  d_6 ,      th_6  ] ]) `

       vv = [1,1,1,1,1,1] 

        variables = [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]
 `       params = [d_1, a_2, d_4, d_6]`
  `      pvals = {d_1:0.6,a_2:0.432,d_4:0.432, d_6:0.4}  # meters`

this just came to the line:

> Trying to find:  th_12
atan2(-r_13*cos(th_4) + r_33*(2*r_13*cos(th_12) + 2*r_33*sin(th_12) + sin(th_4)*cos(th_3)), r_13* (2*r_13*cos(th_12) + 2*r_33*sin(th_12) + sin(th_4)*cos(th_3)) + r_33*cos(th_4))

and then stops there (after 20+ min i stoped the code). 

I also tried one of my own robots with the following DH parameters:

`    if(name == 'SainSmart'): `
`      dh = sp.Matrix([ `
 `     [ sp.pi/2 ,   a_1,  d_1 ,     th_1  ],    `
 `     [   sp.pi ,   a_2,   0 ,      th_2  ],   `
`      [sp.pi/2  ,   a_3,  0 ,      th_3  ],    `
  `    [-sp.pi/2 ,   0,    d_4,       th_4  ],       `
 `     [-sp.pi/2 ,   0,    d_5,       th_5  ], `
`     [       0 ,   0,    0 ,       th_6  ]  ])  `
`    vv = [1,1,1,1,1,1] `

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]
        params = [a_1, a_2, a_3, d_1, d_4, d_5]
        pvals = {a_1:0.3, a_2:1.22, a_3:0.26, d_1:1.1, d_4:1.33, d_5:0.3}  # meters    `

and this does not find the solution for the theta6 joint. 

Am i doing something wrong?

blake5634, 2018-07-26:
There are some robots with no analytic solution and IKBT can't solve those.   There are also SOME robots with solutions which IKBT cannot solve.    It would be great to send us a known solution to your second robot if available.     One case we do know about is if a robot has 3 parallel axes (not yours I think), then we can't yet solve it but we are working on that case.    Thanks for your interest in IKBT.

VidyaPuri, 2018-07-26:
I am sure that there is analytic solution to this robot, but i have not derived one yet. I found two other guys with similar robots (basically the same). One is writen in ruby and one in js. Here are the links to the code.

JavaScript:
> https://github.com/glumb/kinematics/blob/master/src/kinematics.js

Ruby:
> https://github.com/wedesoft/arduino-sainsmart/blob/master/kinematics.rb

the one in Ruby also uses DH parameters and i tried using them with your IKBT, but as said before it did not find solution for theta 6. 

Did i pass the DH parameters wrong way?

blake5634, 2018-07-26:
There are also SOME robots with analytic solutions which IKBT cannot solve.

blake5634, 2018-07-29:
Also, can you clarify that you ARE able to solve the test example robots like  Wrist and Puma?

closed 2018-08-03 by blake5634