uw-biorobotics/IKBT #8: error on running custom Robot arm kinematics

amin101, 2018-07-26:
I added my Robot DH parameters as below:

```
#   The Amin
# 

       
    if(name == 'Amin'):
        dh = sp.Matrix([
            [    th_1   ,    sp.pi/2 ,   0 ,     d_1  ],
            [(sp.pi/2+th_2),    0 ,   d_2   ,     0  ],
            [ th_3 ,   0,    d_3 ,     0  ],   
            [    (sp.pi/2+th_4)   ,   sp.pi/2,  0 ,     0  ],      
            [    th_5    ,   0 ,   0,         (d_4+d_5)    ],      
            [    0    ,   0 ,   0,         0    ]
            ])
        vv = [1,1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5)]
        params = [d_1, d_2, d_3, d_4, d_5]
        pvals = {d_1:1, d_2:1,  d_3:1, d_4:1, d_5:1}  # meters
    
```
after running python ikSolver.py Amin 

```
C:\Users\lion\Desktop\IKBT-master>python ikSolver.py Amin

          Running IK solution

             Working on Amin

kinematics pickle: trying to open  fk_eqns/Amin_pickle.p  in  C:\Users\lion\Desktop\IKBT-master
Did not find VALID stored pickle file:  fk_eqns/Amin_pickle.p
Starting Forward Kinematics
Completed Forward Kinematics
Starting Sum of Angles scan (slow!)
ik_classes: length Robot.mequation_list:  6
[th_1, th_2, th_3, th_4, th_5]
[th_1, th_2, th_3, th_4, th_5]
 Storing results
GOT HERE: robot name:  Amin
Traceback (most recent call last):
  File "ikSolver.py", line 324, in <module>
    R.sum_of_angles_transform(unknowns) #get the sum of angle
  File "C:\Users\lion\Desktop\IKBT-master\ikbtbasics\ik_classes.py", line 306, in sum_of_angles_transform
    th_xy = find_xy(d[thx], d[thy])
  File "C:\Users\lion\Desktop\IKBT-master\ikbtbasics\ik_classes.py", line 132, in find_xy
    thx_s = set(thxy_lookup[thx])
KeyError: d_1 + d_4 + d_5 + th_2 + th_3 + th_4

```

blake5634, 2018-07-26:
Your DH parameters are in the wrong order.   The columns should be alpha(n-1), a(n-1), d(n), theta(n).

closed 2018-07-26 by blake5634

amin101, 2018-07-26:
thanks @blake5634 , I modified my code but still receive error:

```
#   The Amin
# 
# alpha(n-1), a(n-1), d(n), theta(n)
       
    if(name == 'Amin'):
        dh = sp.Matrix([
            [sp.pi/2,    0 ,      d_1,     th_1  ],
            [0,          a_1 ,    0,       th_2  ],
            [0,          a_2,     0,       th_3  ],   
            [sp.pi/2,    0,       0,       th_4  ],      
            [0,          0 ,      d_2,     th_5  ],      
            [0,          0 ,      0,       0     ]
            ])
        vv = [1,1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5)]
        params = [d_1, a_1, a_2,  d_2]
        pvals = {d_1:1, a_1:2,  a_2:4,  d_2:1}  # meters
```
    
error:
```

C:\Users\lion\Desktop\IKBT-master>python ikSolver.py Amin

          Running IK solution

             Working on Amin

kinematics pickle: trying to open  fk_eqns/Amin_pickle.p  in  C:\Users\lion\Desktop\IKBT-master
\Trying to read pre-computed forward kinematics from fk_eqns/Amin_pickle.p
Successfully read pre-computed forward kinematics
GOT HERE: robot name:  Amin
found new 'joint' (sumofangle) variable:  th_45
sumofanglesT: appending  th_45 = th_4 + th_5
th_4 + th_5
Traceback (most recent call last):
  File "ikSolver.py", line 324, in <module>
    R.sum_of_angles_transform(unknowns) #get the sum of angle
  File "C:\Users\lion\Desktop\IKBT-master\ikbtbasics\ik_classes.py", line 306, in sum_of_angles_transform
    th_xy = find_xy(d[thx], d[thy])
  File "C:\Users\lion\Desktop\IKBT-master\ikbtbasics\ik_classes.py", line 132, in find_xy
    thx_s = set(thxy_lookup[thx])
KeyError: th_1 + th_2
```

blake5634, 2018-07-29:
please see issue  #9  (SumOfAngles (KeyError) for custom DH table)