uw-biorobotics/IKBT #12: Unable to solve simple 3-link planar arm

drfastolfe, 2019-02-20:
I wanted to test the solver so I entered a basic 3-link planar arm as a custom robot:
`
        
    if(name == 'Arm_3'):                
        #
        
        dh = sp.Matrix([
        [ 0 , a_1 , 0 , th_1 ],
        [ 0 , a_2 , 0 , th_2 ],
        [ 0 , a_3 , 0 , th_3 ],
        [ 0 , 0 , 0 , 0 ],
        [ 0 , 0 , 0 , 0 ],
        [ 0 , 0 , 0 , 0 ]
        ])

        vv = [1,1,1,0,0,0]
        variables =  [unknown(th_1), unknown(th_2), unknown(th_3)]
        params = [a_1, a_2, a_3]
        pvals = {a_1:1, a_2:1, a_3:1}
`

ikSolver is unable to solve for this arm. It solves for th_1 and th_2 and then gets stuck while solving for th_3. I get no error message, nothing, the terminal is just stuck.

blake5634, 2019-02-23:
We've reproduced your problem.   I have added tests to algebra_solver to expose this issue.   When algebra solver solves a sum of angles eqn like

       th_123 = th_1 + th_2 + th_3
       
then set_solved seems to go into an infinite loop in the solution tree search. 
@dianmuz ???

blake5634, 2019-03-01:
This is another example of same bug from @nithinpachday
BH

if(name == 'Luci'):
# standardize on the order "alpha N-1, a N-1, d N, theta N' for the DH table columns.

    dh = sp.Matrix([
        [sp.pi/2 , 	 0.,         d_1,        th_1],
        [0       , 	 a_1,         0.,         th_2],
        [0       , 	 a_2,         0.,         th_3],
        [-sp.pi/2,   a_3,         0.,         th_4],
        [0       ,	 a_4,         0.,         th_5],
        [0       ,	  0,          0,            0]
        ])
        
    vv = [1, 1, 1, 1, 1, 0]
    variables = [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5)]
    params = [d_1, a_1, a_2, a_3, a_4]
    pvals = {d_1: 0.05, a_1: 0.247, a_2: 0.177, a_3:.085, a_4: 0.}

blake5634, 2019-05-02:
Arm_3 now works with latest master branch.    I'm closing this and opening another issue for  Luci  (which is stuck in "detect_parent" of set_solved.

closed 2019-05-02 by blake5634