uw-biorobotics/IKBT #13: calculation stuck at first unknown

nithinpachday, 2019-02-24:
The calculation was stuck in the first variable of my 5 dof robot.
the dh is as follows,

    if(name == 'Robo'):
        dh = sp.Matrix([
            [sp.pi/2  ,   0 ,  d_1,      th_1  ], 
            [0        ,  a_1,    0,      th_2  ], 
            [0        ,  a_2,    0,      th_3  ],   
            [-sp.pi/2 ,  a_3,    0,      th_4  ],      
            [0        ,  a_4,    0,      th_5  ],
            [0        ,    0,    0,      0  ]
            ])
        vv = [1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5)]
        params = [d_1, a_1, a_2, a_3, a_4]
        pvals = {d_1:0.05,a_1:0.247, a_2:0.177, a_3:0.085, a_4:0}  # meters

blake5634, 2019-02-25:
Make sure vv has 6 values (last one doesn't matter but should be there.)   I'll leave the issue open for a while in case this does not fix it.

nithinpachday, 2019-02-28:
I have updated the robot definition as follows,
if(name == 'Luci'):
        # standardize on the order "alpha N-1, a N-1, d N, theta N' for the DH table columns.
        

        dh = sp.Matrix([
            [sp.pi/2 , 	 0.,         d_1,        th_1],
            [0       , 	 a_1,         0.,         th_2],
            [0       , 	 a_2,         0.,         th_3],
            [-sp.pi/2,   a_3,         0.,         th_4],
            [0       ,	 a_4,         0.,         th_5],
            [0       ,	  0,          0,            0]
            ])
            
        vv = [1, 1, 1, 1, 1, 0]
        variables = [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5)]
        params = [d_1, a_1, a_2, a_3, a_4]
        pvals = {d_1: 0.05, a_1: 0.247, a_2: 0.177, a_3:.085, a_4: 0.}

i have still the issue. Could you please check the output from terminal that i have been given below?
any fixes?

          Running IK solution 

             Working on Luci

Creating a new pickle directory: ./fk_eqns/
kinematics pickle: trying to open  fk_eqns/Luci_pickle.p  in  /home/nithin/Desktop/project_LUCI/documents/IKBT-master
Did not find VALID stored pickle file:  fk_eqns/Luci_pickle.p
Starting Forward Kinematics
Completed Forward Kinematics
Starting Sum of Angles scan (slow!)
ik_classes: length Robot.mequation_list:  8
Starting sum-of-angles scan. Please be patient
 [  1%] [................................................] Sum of Anglesfound new 'joint' (sumofangle) variable: (k= 0 )  th_45
sumofanglesT: appending new equation: th_45 = th_4 + th_5
found new 'joint' (sumofangle) variable: (k= 0 )  th_123
sumofanglesT: appending new equation: th_123 = th_1 + th_2 + th_3
 [  4%] [==..............................................] Sum of Anglesfound new 'joint' (sumofangle) variable: (k= 0 )  th_12
sumofanglesT: appending new equation: th_12 = th_1 + th_2
 [ 13%] [======..........................................] Sum of Anglesfound new 'joint' (sumofangle) variable: (k= 1 )  th_23
sumofanglesT: appending new equation: th_23 = th_2 + th_3
 [100%] [================================================] Sum of Angles
Completed sum-of-angles scan.
[th_1, th_2, th_3, th_4, th_5, th_45, th_123, th_12, th_23]
[th_1, th_2, th_3, th_4, th_5, th_45, th_123, th_12, th_23]
 Storing kinematics pickle for Luci(fk_eqns/Luci_pickle.p)
GOT HERE: robot name:  Luci
Ticking IK BT for  Luci  -------------------------

variable on blackboard: th_1
basenode:  TanID+Solv  ticked 
basenode:  TanID+Solv  FAIL
Looking for unknown:  th_1  in equation:  Py = a_4*sin(th_4) - d_1
Looking for unknown:  th_1  in equation:  -Py - d_1 = -a_4*sin(th_4)
Looking for unknown:  th_1  in equation:  -Py + a_4*r_21*cos(th_5) - a_4*r_22*sin(th_5) - d_1 = 0
variable on blackboard: th_2
basenode:  TanID+Solv  ticked 
basenode:  TanID+Solv  FAIL
Looking for unknown:  th_2  in equation:  Py = a_4*sin(th_4) - d_1
Looking for unknown:  th_2  in equation:  -Py - d_1 = -a_4*sin(th_4)
Looking for unknown:  th_2  in equation:  -Py + a_4*r_21*cos(th_5) - a_4*r_22*sin(th_5) - d_1 = 0
variable on blackboard: th_3
basenode:  TanID+Solv  ticked 
basenode:  TanID+Solv  FAIL
Looking for unknown:  th_3  in equation:  Py = a_4*sin(th_4) - d_1
Looking for unknown:  th_3  in equation:  -Py - d_1 = -a_4*sin(th_4)
Looking for unknown:  th_3  in equation:  -Py + a_4*r_21*cos(th_5) - a_4*r_22*sin(th_5) - d_1 = 0
x2y2 did not find suitable eqns
variable on blackboard: th_4
basenode:  TanID+Solv  ticked 
tan_id: ( th_4 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  a_4    Bw:  -Py - d_1
Cw:  -1    Dw:  r_21*sin(th_5) + r_22*cos(th_5)
tan_id: ( th_4 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  -a_4    Bw:  Py + d_1
Cw:  -1    Dw:  r_21*sin(th_5) + r_22*cos(th_5)
tan_id: ( th_4 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  -1    Bw:  r_21*cos(th_5) - r_22*sin(th_5)
Cw:  -1    Dw:  r_21*sin(th_5) + r_22*cos(th_5)
basenode:  TanID+Solv  FAIL
sincos: checking  th_4

set_solved:  th_4       by:  arcsin
             th_4 = asin((Py + d_1)/a_4) 

 -  - - - - 
[th_1, th_2, th_3, th_4, th_5, th_45, th_123, th_12, th_23]
Trying to find:  th_4
 - - - - - 
asin((Py + d_1)/a_4)
//////////////////////// > 1 sol
curr:  th_4s1
(Py + d_1)/a_4
//////////////////////// > 1 sol
curr:  th_4s2
(Py + d_1)/a_4
finish set_solved th_4
basenode:  Completion Detect  ticked 

           Completion Detector:  9  variables,  1  are solved.
             solved:  th_4 (arcsin);   

basenode:  Completion Detect  FAIL
variable on blackboard: th_5

set_solved:  th_5       by:  algebra
             th_5 = r_21*th_5/sin(th_4 + th_5) 

 -  - - - - 
[th_1, th_2, th_3, th_4, th_5, th_45, th_123, th_12, th_23]
Trying to find:  th_5
 - - - - - 
r_21*th_5/sin(th_4 + th_5)

blake5634, 2019-03-01:
Thanks - this is same as Issue #12 but a 2-way version , thanks for this test case.   I'm going to close this but please follow Issue #12. 
BH

closed 2019-03-01 by blake5634