uw-biorobotics/IKBT #14: Index out of range

ekorudiawan, 2019-04-09:
I try to create a simple robot but I got error
```python
if(name == 'Contoh'):  #Testing robot
        dh = sp.Matrix([
            [    0    ,    0 ,   0 ,     th_1  ],
            [    0    ,   a_1,   0 ,     th_2  ],
            [    0    ,   a_2,   0 ,     th_3  ]
            ])
        vv = [1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3)]
        params = [a_1,a_2]
        pvals = {a_1: .425, a_2: .392}  # meters
```
Error message

```bash
kinematics pickle: trying to open  fk_eqns/Contoh_pickle.p  in  D:\Source-Code\Robotics\IKBT
Did not find VALID stored pickle file:  fk_eqns/Contoh_pickle.p
Starting Forward Kinematics
Traceback (most recent call last):
  File "ikSolver.py", line 108, in <module>
    [M, R, unknowns] = kinematics_pickle(robot, dh, params, pvals, vv, unknowns, testing)
  File "D:\Source-Code\Robotics\IKBT\ikbtbasics\ik_classes.py", line 91, in kinematics_pickle
    m.forward_kinematics()
  File "D:\Source-Code\Robotics\IKBT\ikbtbasics\kin_cl.py", line 155, in forward_kinematics
    self.T_34 = Link_S(self.DH[3,al], self.DH[3,a], self.DH[3,d], self.DH[3,th])
  File "C:\Users\ekoru\Anaconda3\envs\ikbt\lib\site-packages\sympy\matrices\dense.py", line 122, in __getitem__
    return self.extract(i, j)
  File "C:\Users\ekoru\Anaconda3\envs\ikbt\lib\site-packages\sympy\matrices\common.py", line 333, in extract
    rowsList = [a2idx(k, self.rows) for k in rowsList]
  File "C:\Users\ekoru\Anaconda3\envs\ikbt\lib\site-packages\sympy\matrices\common.py", line 2304, in a2idx
    raise IndexError("Index out of range: a[%s]" % (j,))
IndexError: Index out of range: a[3]
```

blake5634, 2019-04-11:
Make sure you have 6 rows in the DH table.    Just pad the sp.Matrix (as you show above) with 3 more rows of   [0,0,0,0]

BH

ekorudiawan, 2019-04-17:
> Make sure you have 6 rows in the DH table. Just pad the sp.Matrix (as you show above) with 3 more rows of [0,0,0,0]
> 
> BH

Still get same error,, I think this value should be change also 

```python
vv = [1,1,1]
```

How to change this one?

blake5634, 2019-04-18:
Extend that to 6 values as well.  Also extend the "variables =" statement.   Please see the video https://youtu.be/hXzY5vrvWkU

closed 2019-04-18 by blake5634