uw-biorobotics/IKBT #15: bad notation collection

michaeliskndr, 2019-04-23:
Hi there,

I tried to solve my own robots configuration for months and keep failing. 
And i found this. Really interesting and helpful.

When i tried to solved with this, it gave a results like this:

```
variable on blackboard: th_1
basenode:  Algebra ID  ticked 

running:  Algebra ID
 eqns w/ one unk:  0
 eqns w/ two unk:  5
basenode:  Algebra ID  FAIL
basenode:  TanID+Solv  ticked 
basenode:  TanID+Solv  FAIL
variable on blackboard: th_2
basenode:  Algebra ID  ticked 

running:  Algebra ID
 eqns w/ one unk:  0
 eqns w/ two unk:  5
basenode:  Algebra ID  FAIL
basenode:  TanID+Solv  ticked 
basenode:  TanID+Solv  FAIL
variable on blackboard: th_3
basenode:  Algebra ID  ticked 

running:  Algebra ID
 eqns w/ one unk:  0
 eqns w/ two unk:  5
basenode:  Algebra ID  FAIL
basenode:  TanID+Solv  ticked 
tan_id: ( th_3 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  -l_3    Bw:  -Py + d_6*r_23 - l_2
Cw:  sin(th_4)    Dw:  -r_23
tan_id: ( th_3 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  -l_3    Bw:  -Py + d_6*r_23 - l_2
Cw:  -sin(th_4)    Dw:  r_23
tan_id: ( th_3 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  l_3    Bw:  Py - d_6*r_23 + l_2
Cw:  sin(th_4)    Dw:  -r_23
tan_id: ( th_3 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  l_3    Bw:  Py - d_6*r_23 + l_2
Cw:  -sin(th_4)    Dw:  r_23
tan_id: ( th_3 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  1    Bw:  r_21*sin(th_5) + r_22*cos(th_5)
Cw:  sin(th_4)    Dw:  -r_23
tan_id: ( th_3 )   0 =  Aw*sin(th)+Bw , 0 = Cw*cos(th) + Dw 
Aw:  1    Bw:  r_21*sin(th_5) + r_22*cos(th_5)
Cw:  -sin(th_4)    Dw:  r_23
basenode:  TanID+Solv  FAIL
x2y2 did not find suitable eqns
variable on blackboard: th_4
basenode:  Algebra ID  ticked 

running:  Algebra ID
 eqns w/ one unk:  0
 eqns w/ two unk:  5
basenode:  Algebra ID  FAIL
basenode:  TanID+Solv  ticked 
basenode:  TanID+Solv  FAIL
variable on blackboard: th_5
basenode:  Algebra ID  ticked 

running:  Algebra ID
 eqns w/ one unk:  0
 eqns w/ two unk:  5
basenode:  Algebra ID  FAIL
[... 411 lines cut ...]

```
if(name == 'ArmRobo'):
        # standardize on the order "alpha N-1, a N-1, d N, theta N' for the DH table columns.
        
        # Olson 2013
        # DOF: 6
        # methods to test: m5, m3, 
        # Yb = d_1, Xb = d_2, L1 = l3, L2 = l4, L3 = l5
        dh = sp.Matrix([
            [sp.pi/2,    0.,       l_2,              th_1],
            [0,         l_1,        0.,              th_2],
            [sp.pi/2,    0.,        0.,  (th_3-(sp.pi/2))],
            [-sp.pi/2,   0.,       l_3,              th_4],
            [sp.pi/2,    0.,        0.,              th_5],
            [0,          0.,       d_6,       0.]
            ])
            
        vv = [1, 1, 1, 1, 1, 0]
        variables = [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(d_6)]
        params = [l_1, l_2, l_3]
        pvals = {l_1: 0.19681, l_2: 0.251, l_3:0.145423}
```

Big thanks for your help, really appreciate it.

blake5634, 2019-04-27:
Thanks for submitting.   I've reproduced your bug and we'll try to take a deep look.

michaeliskndr, 2019-06-12:
thank you mr blake for the updates, it's exciting to wait for your further updates

blake5634, 2019-07-05:
An update on this bug:    Remember that a certain set of DH tables do not have an analytic solution.   Another set of DH tables are not solvable by IKBT.  Ideally these sets are the same but I can't prove it.  

The latest "testing" branch has streamlined the BT for this and similar cases in which none of our solvers can solve any of the variables at all.      We no print  that we are giving up instead of "bad notation collection".    We have also made some efficiency improvements in the BT and other cleanups. 

All tests currently pass in the "testing branch" commit c92570b61a32

closed 2019-07-05 by blake5634