uw-biorobotics/IKBT #20: Error in Python CodeGen: global name 'acos' is not defined

xoxota99, 2019-05-08:
If the set of solution methods contains the arccos or arcsin methods, the correct imports are not generated into the resulting python code, resulting in the below errors:
- global name 'acos' is not defined
- global name 'asin' is not defined

blake5634, 2019-05-09:
I think I recall that bug, just do    acos--> np.acos      asin---> np.asin
I'll keep issue open -- it's an easy fix.  (but also easy to fix in your generated code   ;)   )

xoxota99, 2019-05-09:
Yep, super easy fix. Can also just 'from math import acos'

xoxota99, 2019-05-09:
@blake5634 is there a support forum for IKBT, or a better place for me to ask general-knowledge questions about the project, besides the Github issues queue?

blake5634, 2019-05-11:
There is not yet a support forum.   I don't mind using the github issues for now but what do you think would be the best platform for the newbie questions, feature requests, etc???

xoxota99, 2019-05-12:
Github issues works great, I just have a lot of newbie questions, and don't want to clog up the wrong inbox :-)

xoxota99, 2019-05-20:
Fixed in https://github.com/uw-biorobotics/IKBT/commit/e1517a553acc8687dda01d5da5dacf8f29c9fe3f . Closing.

closed 2019-05-20 by xoxota99