uw-biorobotics/IKBT #21: IkSolver infinite loop / hang for Khat6DOF

xoxota99, 2019-05-10:
Running 'python ikSolver.py Khat6DOF', script appears to enter an infinite loop.

Last output:
`
set_solved:  th_23       by:  simultaneous eqn
             th_23 = atan2((-Pz + l_1)*(-a_2*cos(th_3) - a_3) - (a_2*sin(th_3) + l_4)*(-Px*cos(th_1) - Py*sin(th_1) + a_1*cos(th_1)), (-Pz + l_1)*(a_2*sin(th_3) + l_4) + (-a_2*cos(th_3) - a_3)*(-Px*cos(th_1) - Py*sin(th_1) + a_1*cos(th_1)))

atan2((-Pz + l_1)*(-a_2*cos(th_3) - a_3) - (a_2*sin(th_3) + l_4)*(-Px*cos(th_1) - Py*sin(th_1) + a_1*cos(th_1)), (-Pz + l_1)*(a_2*sin(th_3) + l_4) + (-a_2*cos(th_3) - a_3)*(-Px*cos(th_1) - Py*sin(th_1) + a_1*cos(th_1)))
`

Followed by forty minutes of waiting until I Ctrl-C'ed out of there.

Stack trace (following Ctrl-C):
`
Traceback (most recent call last):
  File "ikSolver.py", line 339, in <module>
    ikbt.tick("Test a full solver", bb)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/behaviortree.py", line 120, in tick
    state = self.root._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/decorators/repeatuntilsuccess.py", line 20, in tick
    status = self.child._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/composites/sequence.py", line 14, in tick
    status = node._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/decorators/repeatuntilsuccess.py", line 20, in tick
    status = self.child._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/composites/sequence.py", line 14, in tick
    status = node._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/composites/priority.py", line 13, in tick
    status = node._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/composites/sequence.py", line 14, in tick
    status = node._execute(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/ikbtleaves/two_eqn_m7.py", line 161, in tick
    curr_unk.set_solved(R, unknowns)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/ikbtbasics/kin_cl.py", line 190, in set_solved
    curr_node.detect_parent(R)
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/ikbtbasics/solution_graph_v2.py", line 130, in detect_parent
    if par != other_par and related(par, other_par):
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/ikbtbasics/solution_graph_v2.py", line 75, in related
    if curr == end_node:
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/ikbtbasics/solution_graph_v2.py", line 106, in __eq__
    return self.symbol == other.symbol
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/lib/python2.7/site-packages/sympy/core/basic.py", line 341, in __eq__
    return self._hashable_content() == other._hashable_content()
  File "/Users/philippd/Google Drive/projects/flexo/IKBT/lib/python2.7/site-packages/sympy/core/symbol.py", line 252, in _hashable_content
    return (self.name,) + tuple(sorted(self.assumptions0.items()))
KeyboardInterrupt
`

xoxota99, 2019-05-10:
Sorry, GitHub doesn't like formatting code, apparently (ironically).

blake5634, 2019-05-12:
Current work is focused in on this bug via a general refactor of solution_graph.

closed 2019-05-12 by blake5634