uw-biorobotics/IKBT #22: Help: Am I doing this right?

xoxota99, 2019-05-17:
I've defined what I think is my 6DOF robot arm (Freebody diagram [here](https://imgur.com/a/0cG1Nq3)), but not 100% sure I'm putting the right DH Params in the right place. I'm also not sure that the signs are correct on all my params. For example: 

1. running this through ik_solver, there is no solution for joint #6.
2. Not sure how to specify the tool work frame (As a cartesian / Euler coordinate relative to J6 work frame).

The Code:
```
    if(name == 'Flexo'):
        # standardize on the order "alpha N-1, a N-1, d N, theta N' for the DH table columns.

        dh = sp.Matrix([
            [-sp.pi/2,        l_1,         d_1,        th_1],
            [0,               l_2,         0.,         th_2],
            [sp.pi/2,         l_3,         0.,         th_3],
            [-sp.pi/2,        0.,          d_4,        th_4],
            [sp.pi/2,         0.,          0.,         th_5],
            [0,               0.,          d_6,        th_6]
        ])

        vv = [1, 1, 1, 1, 1, 1]
        variables = [unknown(th_1), 
                    unknown(th_2), 
                    unknown(th_3), 
                    unknown(th_4), 
                    unknown(th_5), 
                    unknown(th_6)]
        params = [l_1, l_2, l_3, d_1, d_4, d_6]
        pvals = {l_1: 90, l_2: 250, l_3: 66, d_1: 140, d_4: -340, d_6: -60}     #in millimeters
```

blake5634, 2019-05-18:
Thanks for posting.      regarding your points: 1) you have 6 non-zero length offsets (li's and di's) and this suggest that you MAY have a robot with no solution (but I can't prove it).   2)  I'm working on a complete re-write of the solution set part -- even though it works great for some tough robots like Puma, it still has gaps.  However there is no meaning to the solution set if all joints are not solved. 

Regarding your other points  assigning correct DH parameters is tricky and it's unfortunately beyond our scope here.    for point 3)  transform you desired pose to the base frame of the robot (i.e. where it is mounted) and invert the tool offset and apply it to the desired pose..

xoxota99, 2019-05-20:
Thanks Blake. I agree this may be more of a problem with my understanding of DH params, and not the software. This robot arm is a pretty standard geometry (similar to e.g. [ABB IRB-20](https://new.abb.com/products/robotics/industrial-robots/irb-120)), so if my parameters look "weird", it's probably because I got them wrong (despite all the many YouTube videos I've watched about DH Params).

xoxota99, 2019-05-20:
In desperation, I tried copying the configuration for the Kuka KR-60 ("Khat6DOF"). Unfortunately, that configuration seems to result in some kind of infinite loop ( See #15).

blake5634, 2019-07-05:
I'll close these - the Khat6DOF infinite loop is covered in #18 which will remain open.

closed 2019-07-05 by blake5634