uw-biorobotics/IKBT #24: th_* variables undeclared, "argument" variable multi-assigned in generated cpp code

xoxota99, 2019-05-21:
Generated cpp code appears to assign results of multiple solutions to the same "argument" variable, and uses undeclared th_1, th_2, th_3 (etc) variables in solution calculations. The code won't compile, and is unusable.

The generated python code doesn't appear to have these issues.

My robot definition:
```
    if(name == 'Flexo'):
       # standardize on the order "alpha N-1, a N-1, d N, theta N' for the DH table columns.

        a_1 = sp.symbols("a_1")

        dh = sp.Matrix([                 
            [ 0,         0,      d_1 ,     th_1  ],
            [ -sp.pi/2,  a_1,    0,        th_2-sp.pi/2 ],
            [ 0,         a_2,    0,        th_3  ],   
            [ -sp.pi/2,  a_3,    d_4,      th_4  ],      
            [ sp.pi/2,   0,      0,        th_5  ],
            [ -sp.pi/2,  0,      0,        th_6  ]
        ])
        vv = [1,1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]

        params = [a_1, d_1, a_2, a_3, d_4]
        # all values in mm.
        pvals = {a_1: 90.11, 
            d_1: 140.431, 
            a_2: 250, 
            a_3: -66.306, 
            d_4: 342.451}
```

xoxota99, 2019-05-21:
The generated cpp code does things like this:

Assigning "argument" variable multiple times in a row without using it:
```
    // solution 0
    // solvemethod: simultaneous eqn
    //    argument: abs(-(-Pz + d_1)*(a_2*sin(th_3) - d_4) + (-a_2*cos(th_3) - a_3)*(-Px*cos(th_1) - Py*sin(th_1) + a_1)) + abs((-Pz + d_1)*(-a_2*cos(th_3) - a_3) + (a_2*sin(th_3) - d_4)*(-Px*cos(th_1) - Py*sin(th_1) + a_1))
    // simultaneous equations - based solution:
     argument = abs(-(-Pz + d_1) * (a_2 * sin(th_3) - d_4) + (-a_2 * cos(th_3) - a_3) * (-Px * cos(th_1s1) - Py * sin(th_1s1) + a_1)) + abs((-Pz + d_1) * (-a_2 * cos(th_3) - a_3) + (a_2 * sin(th_3) - d_4) * (-Px * cos(th_1s1) - Py * sin(th_1s1) + a_1));

    // solution 1
    // solvemethod: simultaneous eqn
    //    argument: abs(-(-Pz + d_1)*(a_2*sin(th_3) - d_4) + (-a_2*cos(th_3) - a_3)*(-Px*cos(th_1) - Py*sin(th_1) + a_1)) + abs((-Pz + d_1)*(-a_2*cos(th_3) - a_3) + (a_2*sin(th_3) - d_4)*(-Px*cos(th_1) - Py*sin(th_1) + a_1))
    // simultaneous equations - based solution:
    argument = abs(-(-Pz + d_1) * (a_2 * sin(th_3) - d_4) + (-a_2 * cos(th_3) - a_3) * (-Px * cos(th_1s2) - Py * sin(th_1s2) + a_1)) + abs((-Pz + d_1) * (-a_2 * cos(th_3) - a_3) + (a_2 * sin(th_3) - d_4) * (-Px * cos(th_1s2) - Py * sin(th_1s2) + a_1));

    // solution 2
    // solvemethod: simultaneous eqn
    //    argument: abs(-(-Pz + d_1)*(a_2*sin(th_3) - d_4) + (-a_2*cos(th_3) - a_3)*(-Px*cos(th_1) - Py*sin(th_1) + a_1)) + abs((-Pz + d_1)*(-a_2*cos(th_3) - a_3) + (a_2*sin(th_3) - d_4)*(-Px*cos(th_1) - Py*sin(th_1) + a_1))
    // simultaneous equations - based solution:
    argument = abs(-(-Pz + d_1) * (a_2 * sin(th_3) - d_4) + (-a_2 * cos(th_3) - a_3) * (-Px * cos(th_1s2) - Py * sin(th_1s2) + a_1)) + abs((-Pz + d_1) * (-a_2 * cos(th_3) - a_3) + (a_2 * sin(th_3) - d_4) * (-Px * cos(th_1s2) - Py * sin(th_1s2) + a_1));

    // solution 3
    // solvemethod: simultaneous eqn
    //    argument: abs(-(-Pz + d_1)*(a_2*sin(th_3) - d_4) + (-a_2*cos(th_3) - a_3)*(-Px*cos(th_1) - Py*sin(th_1) + a_1)) + abs((-Pz + d_1)*(-a_2*cos(th_3) - a_3) + (a_2*sin(th_3) - d_4)*(-Px*cos(th_1) - Py*sin(th_1) + a_1))
    // simultaneous equations - based solution:
    argument = abs(-(-Pz + d_1) * (a_2 * sin(th_3) - d_4) + (-a_2 * cos(th_3) - a_3) * (-Px * cos(th_1s1) - Py * sin(th_1s1) + a_1)) + abs((-Pz + d_1) * (-a_2 * cos(th_3) - a_3) + (a_2 * sin(th_3) - d_4) * (-Px * cos(th_1s1) - Py * sin(th_1s1) + a_1));

blake5634, 2019-05-22:
Thanks for these.    C++ output is the least tested part of IKBT. We certainly released before the code gen could "just run".    I suspect that the issue you identify with th_x will require the solution graph refactor that I'm pushing on now.

xoxota99, 2019-05-22:
It sounds like you're already working on it a ton. Do you want me to continue posting issues as I find them? OR just wait for the next release?

blake5634, 2019-07-05:
Yes - pls post any issues you find.    Probably all C++ output issues should stay in this one (#24). THX