uw-biorobotics/IKBT #25: RRR articulated manipulator

BradleyCorbett, 2019-06-18:
I was trying to solve a simple 3 dof arm but was coming into an issue very similar to the one with the 3 link planar arm where it wouldn't solve for theta_3

[image]

these are the DH values I used to create it in the robots file.

     `    if(name == 'Brad'):    # 3-RRR articulated manipulator
        
        # Brad 2019
        # DOF: 3
        dh = sp.Matrix([
            [sp.pi/2, 0., d_1, th_1],
            [0., l_2, 0., th_2],
            [0., l_3, 0., th_3],
            [0., 0., 0., 0.],
            [0., 0., 0., 0.],
            [0., 0., 0., 0.]
            ])
            
        vv = [1, 1, 1, 0, 0, 0]
        variables = [unknown(th_1), unknown(th_2), unknown(th_3)]
        params = [d_1, l_2, l_3]
        pvals = {d_1: 0.06, l_2: 0.15, l_3: 0.15}`

 I'm probably missing something really simple but didn't know how else to ask for your help. thank you very much in advance.

blake5634, 2019-06-18:
Hi Brad,
    Your DH parameters don't correspond to the schematic, the alpha_n-1 = sp.pi/2 should be in row 2.   This gives a reasonable result.    Note we have a bug where th_2 and th_3 are known but it uses atan2() to solve th_23 (!)   we're working on this but th_23 should be ignored anyway since it's only needed for intermediate work.

closed 2019-06-18 by blake5634

BradleyCorbett, 2019-06-19:
that's strange as I took the dh off of a research paper. so I changed the dh to this 

`dh = sp.Matrix([
            [0., 0., d_1, th_1],
            [sp.pi/2, l_2, 0., th_2],
            [0., l_3, 0., th_3],
            [0., 0., 0., 0.],
            [0., 0., 0., 0.],
            [0., 0., 0., 0.]
            ])`

now the program gives me the python code and gives the solution as 
` solution_list.append( [  th_1,  th_23,  th_2s1,  th_3s1,  ] )`

so I ignore th_23 and just take the three remaining theta values?
and then the positioning of the end effector you have to give it x,y z and an angle? sorry for all the questions I'm just a little lost. thanks very much