uw-biorobotics/IKBT #30: Failed to solve UR5

VidyaPuri, 2020-01-03:
I tried the ikSolver.py with UR5 and it encountered an error

NameError: global name 'find_xy' is not defined

is this something known? I did not touch the solutions just tried the Wrist and then wanted to do UR5 but it stopped after finishing FK.

capaulson, 2020-01-28:
@VidyaPuri same issue. This isn't an issue under the branch ReleaseVersion, but that also doesn't provide a solution for the UR5. I'm also trying with some slightly updated parameters for the UR5e and the result is the same. @blake5634 any pointers on how to solve this?

blake5634, 2020-01-31:
I'll try to find some time to dig into this.   I recall that we got UR-3 working a year or so ago and I thought UR3 and UR5 had the same DH table.    Can anyone repeat this bug with UR3?

blake5634, 2020-02-04:
UR5 is now working again and Latex output is cleaner.    Solution sets are very numerous and have not been validated yet.   See branch RepairUR5Regression

closed 2020-02-04 by blake5634

capaulson, 2020-02-05:
Just a quick follow up, I had to add: 
```
from solution_graph_v2 import *
```
to the file ikbtbasic/kin_cl.py. I'm not sure this is the correct thing to do, but it solves a few runtime errors around not being able to find `Node` and `Edge`. 

Thanks for the fix. Just out of curiosity, how do you validate results?

blake5634, 2020-02-05:
Funny, I can't reproduce that bug.   (i.e. I don't need to add anything to RepairUR5Regression.
For validation results see our paper (cited / linked in the README.md).