uw-biorobotics/IKBT #32: Sum of angles cos(x-y) error

fredrikryden, 2020-03-18:
Hi @blake5634, great to see this project on github. 

I tried to use it for one of the Kinova arms but got this failure:
`Starting sum-of-angles scan. Please be patient
 [  1%] [................................................] Sum of AnglesTraceback (most recent call last):
  File "ikSolver.py", line 108, in <module>
    [M, R, unknowns] = kinematics_pickle(robot, dh, params, pvals, vv, unknowns, testing)
  File "/home/fredrik/src/IKBT/ikbtbasics/ik_classes.py", line 101, in kinematics_pickle
    R.sum_of_angles_transform(unks)  # find sum of angles
  File "/home/fredrik/src/IKBT/ikbtbasics/ik_classes.py", line 320, in sum_of_angles_transform
    rhs, newj, newe= sum_of_angles_sub(self, rhs, variables)
  File "/home/fredrik/src/IKBT/ikbtbasics/ik_classes.py", line 387, in sum_of_angles_sub
    nil.append( str(get_variable_index(variables, v)) )
  File "/home/fredrik/src/IKBT/ikbtbasics/ik_classes.py", line 442, in get_variable_index
    assert found, 'Error: trying to get index of an unknown joint variable: ' + str(symb)
AssertionError: Error: trying to get index of an unknown joint variable: -th_3
`

I assume it has to do with the fact the equations (which I partially solved by hand) involve cos(x-y) and thus it tries to find '-th_3' instead of 'th_3'. Would love to help fix this problem but not sure exactly where that fix would go. 

 if(name == 'Jaco'):  # j2s6s300
        dh = sp.Matrix([
            [sp.pi, 0,  d_0,     th_1],
            [sp.pi/2, 0,  0,     th_2],
            [sp.pi, a_2,  d_2,     th_3],
            [sp.pi/2, 0,  d_3,     th_4],
            [-sp.pi/2, 0,  0,     th_5],
            [-sp.pi/2, 0,  d_5,     th_6]
        ])
        vv = [1, 1, 1, 1, 1, 1]
        variables = [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]
        params = [d_0, d_2, a_2, d_3, d_5]
        pvals = {d_0: -0.27545, d_2: -0.0114, a_2: 0.410,
                 d_3: -0.31105, d_5: .10375}  # meters

blake5634, 2020-03-18:
Hi Frederik, How are things going?   I'll try to take a look at the -th_3 issue.   Normally IKBT should totally handle a negative value like that.    I'll have to do a little digging. 
Blake

blake5634, 2020-03-18:
Hi Fredrik!
   How are you?  (we're locked down here but doing well - no symptoms
anyway).    UW has switched entirely to online classes for the foreseeable
future.  Quite a new world all of a sudden!

   I'll try to take a look at the -th_3 issue.   Normally IKBT should
totally handle a negative value like that.    I'll have to do a little
digging.

By the way, how do you like the Kinova robots?  are they as good as UR or
Franke?
Blake

fredrikryden, 2020-03-24:
@blake5634, I sent you an email on your university email.

Related to this particular robot, what about the approach of sliding frames along their axis of rotation. Like how you can shift the offset along z between the two parallel axes. Or you can slide the first or last frame along its own axis (for instance this can eliminat d_5). 

I guess the question is, is sliding the frames a valid approach to simplifying the IK or is it just a sign that I assign my frames poorly to begin with?

Fredrik

closed 2023-01-31 by blake5634