uw-biorobotics/IKBT #33: Stuck in a loop while solving th_5

Belinbelin, 2020-04-11:
I tried to use the ik_solver for a 5DOF robot with given D-H parameters:
[-pi/2      0     l1     th_1]
[0            l_2  0      th_2]
[pi/2       0     0      th_3]
[-pi/2      0     0      th_4]
[0            l_4   0      th_5]
[0            0      0      0     ]

The solver finds 2 variables and then get stuck in a loop finding th_5.
Is it possible that there is simply no analytical solutions ?

Tried the program on differents robots, it works very well and output the solutions in a very pleasant way to use them.

blake5634, 2020-04-11:
Yes,  while there is a famous theorem (Piper and Roth) about sufficient condition for solvability,  I don't know of a necessary condition (i.e. that you can apply to the DH params that show in advance there is no analytic solution).  If there is a necessary condition, it would be really cool to apply that to the user's DH params and just print out "Sorry there is no analytic solution to this robot".

blake5634, 2020-05-25:
oops - forgot to close this back in April.

closed 2020-05-25 by blake5634