uw-biorobotics/IKBT #38: find_xy error

shbang91, 2020-12-30:
I've tried to implement IKBT on my personal robot, but I found the following error:

```
Assigner: variable on blackboard: th_1
Traceback (most recent call last):
  File "ikSolver.py", line 346, in <module>
    ikbt.tick("Test a full solver", bb)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/behaviortree.py", line 120, in tick
    state = self.root._execute(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/decorators/repeatuntilsuccess.py", line 20, in tick
    status = self.child._execute(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/composites/sequence.py", line 14, in tick
    status = node._execute(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/decorators/repeatuntilsuccess.py", line 20, in tick
    status = self.child._execute(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/composites/sequence.py", line 14, in tick
    status = node._execute(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 128, in _execute
    status = self._tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/b3/core/basenode.py", line 153, in _tick
    status = self.tick(tick)
  File "/Users/seunghyeonbang/Repository/IKBT/ikbtleaves/sum_transform.py", line 95, in tick
    th_xy = find_xy(d[thx], d[thy])
  File "/Users/seunghyeonbang/Repository/IKBT/ikbtbasics/ik_classes.py", line 137, in find_xy
    thx_s = set(thxy_lookup[thx])
KeyError: th_34
```

Before I got this error message, IBKT package did not find the function find_xy and I comment in the function definition in ik_classes.py. But, I believe it still has an issue inside of that find_xy function. Do you have any suggestion?

blake5634, 2020-12-30:
Please send us the IKBT robot description so we can try to reproduce the error.  THX.

shbang91, 2020-12-30:
Here is my robot description file(ik_robots.py). My robot name is atlas_left_leg. Thank you for your support!
`#!/usr/bin/python
#
#     Inverse Kinematics Classes
#

# Copyright 2017 University of Washington

# Developed by Dianmu Zhang and Blake Hannaford 
# BioRobotics Lab, University of Washington

# Redistribution and use in source and binary forms, with or without modification, are permitted provided that the following conditions are met:

# 1. Redistributions of source code must retain the above copyright notice, this list of conditions and the following disclaimer.

# 2. Redistributions in binary form must reproduce the above copyright notice, this list of conditions and the following disclaimer in the documentation and/or other materials provided with the distribution.

# 3. Neither the name of the copyright holder nor the names of its contributors may be used to endorse or promote products derived from this software without specific prior written permission.

# THIS SOFTWARE IS PROVIDED BY THE COPYRIGHT HOLDERS AND CONTRIBUTORS "AS IS" AND ANY EXPRESS OR IMPLIED WARRANTIES, INCLUDING, BUT NOT LIMITED TO, THE IMPLIED WARRANTIES OF MERCHANTABILITY AND FITNESS FOR A PARTICULAR PURPOSE ARE DISCLAIMED. IN NO EVENT SHALL THE COPYRIGHT HOLDER OR CONTRIBUTORS BE LIABLE FOR ANY DIRECT, INDIRECT, INCIDENTAL, SPECIAL, EXEMPLARY, OR CONSEQUENTIAL DAMAGES (INCLUDING, BUT NOT LIMITED TO, PROCUREMENT OF SUBSTITUTE GOODS OR SERVICES; LOSS OF USE, DATA, OR PROFITS; OR BUSINESS INTERRUPTION) HOWEVER CAUSED AND ON ANY THEORY OF LIABILITY, WHETHER IN CONTRACT, STRICT LIABILITY, OR TORT (INCLUDING NEGLIGENCE OR OTHERWISE) ARISING IN ANY WAY OUT OF THE USE OF THIS SOFTWARE, EVEN IF ADVISED OF THE POSSIBILITY OF SUCH DAMAGE.
import sympy as sp
#import numpy as np
from ikbtbasics.kin_cl import *
from ikbtfunctions.helperfunctions import *
from ikbtbasics.ik_classes import *     # special classes for Inverse kinematics in sympy
#

####
#
#   NOTE: due to an obscure sympy bug, you cannot use numerical values in any DH position.   Use a symbolic constant
#         instead (like a_3 below), declare it in params, and give it your value in pvals
#
#####
def robot_params(name):
    pvals = {}   # null for most robots
    List = ['UR5', 'Puma', 'Chair_Helper', 'Brad', 'ArmRobo', 'Wrist', 'Arm_3', 'MiniDD', 'Olson13','Stanford', 'Chair6DOF','Khat6DOF','Craig417','atlas_left_leg']
    if not (name in List):
        print 'robot_params(): Unknown robot, ' + name 
        print 'Here are the defined robots: '
        for n in List:
            print '   ', n
        quit()
        
    if(name == 'ArmRobo'):    # submitted by issue #15

        # standardize on the order "alpha N-1, a N-1, d N, theta N' for the DH table columns.
        
        # Olson 2013
        # DOF: 6
        # methods to test: m5, m3, 
        # Yb = d_1, Xb = d_2, L1 = l3, L2 = l4, L3 = l5
        dh = sp.Matrix([
            [sp.pi/2,    0.,       l_2,              th_1],
            [0,         l_1,        0.,              th_2],
            [sp.pi/2,    0.,        0.,              th_3],
            [-sp.pi/2,   0.,       l_3,              th_4],
            [sp.pi/2,    0.,        0.,              th_5],
            [0,          0.,       d_6,                0.]
            ])
            
        vv = [1, 1, 1, 1, 1, 0]
        variables = [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(d_6)]
        params = [l_1, l_2, l_3]
        pvals = {l_1: 0.19681, l_2: 0.251, l_3:0.145423}
#Big thanks for your help, really appreciate it.      

    if(name == 'Arm_3'):         # two link planar test (Issue #12) 
        #
        dh = sp.Matrix([
        [ 0 , a_1 , 0 , th_1 ],
[... 396 lines cut ...]
        dh = sp.Matrix([                  ##  This one requires sum-of-angles.
        [  0,       a_1 , 0. ,     th_1  ],
        [ sp.pi/2,    0.,  0.  ,      th_2  ],
        [ -sp.pi/2,  a_2,  l_1  ,      th_3  ],   
        [ 0 , a_3, l_2,       th_4  ],      
        [ 0 , a_4  ,  0. ,       th_5  ],
        [ sp.pi/2 ,   0., l_3  ,       th_6  ]
        ])
        vv = [1,1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]

        params = [a_1, a_2, l_1, a_3, l_2, a_4,l_3]
        pvals = {a_1: 0.089, a_2:0.66, l_1:0.05,  a_3:0.377327, l_2:0.0225, a_4:0.422, l_3:0.003}   
        
        
    ################## (all robots) ######################
    ##  make sure each unknown knows its position (index)
    i = 1
    for v in variables:   
        v.n = i
        i+=1
        
    return [dh, vv, params, pvals, variables]
   `

shbang91, 2020-12-30:
Here is my robot description file(ik_robots.py). My robot name is atlas_left_leg. Thank you for your support!
In case you need the whole file, I send you the ik_robots.py to your email.
```
if(name == 'atlas_left_leg'):                
        
        dh = sp.Matrix([                  ##  This one requires sum-of-angles.
        [  0,       a_1 , 0. ,     th_1  ],
        [ sp.pi/2,    0.,  0.  ,      th_2  ],
        [ -sp.pi/2,  a_2,  l_1  ,      th_3  ],   
        [ 0 , a_3, l_2,       th_4  ],      
        [ 0 , a_4  ,  0. ,       th_5  ],
        [ sp.pi/2 ,   0., l_3  ,       th_6  ]
        ])
        vv = [1,1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]

        params = [a_1, a_2, l_1, a_3, l_2, a_4,l_3]
        pvals = {a_1: 0.089, a_2:0.66, l_1:0.05,  a_3:0.377327, l_2:0.0225, a_4:0.422, l_3:0.003}   
     
```

blake5634, 2020-12-30:
Sorry, your source code is mangled by  email.   Please attach it as a .py
file, not mousing into the message.
THX

closed 2023-01-31 by blake5634