uw-biorobotics/IKBT #39: Implementing my own arm

Ipuch, 2021-06-17:
Hi,

I'd like to use your inverse kinematic solver for my own robot. Here is the parameters in DH formalism.

DHRobot: KINOVA, 6 joints (PPRRRR), standard DH parameters
┏━━━━━━━┳━━━━━━━━━━┳━━━━━━┳━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━┓
┃  θⱼ           ┃    dⱼ               ┃  aⱼ        ┃         ⍺ⱼ                                 ┃   q⁻             ┃   q⁺          ┃
┣━━━━━━━╋━━━━━━━━━━╋━━━━━━╋━━━━━━━━━━━━━━━━━━━━━╋━━━━━━━━━╋━━━━━━━━┫
┃0.0°         ┃       q1           ┃    0      ┃              -90.0°                      ┃   -0.33         ┃   0.33       ┃
┃-90.0°      ┃       q2           ┃    0      ┃               90.0°                      ┃   -0.33         ┃   0.33       ┃
┃ q3           ┃      0.1           ┃    0      ┃               90.0°                      ┃ -180.0°       ┃ 180.0°      ┃
┃ q4           ┃     0.01         ┃ 0.25     ┃ 29.999999999999996°        ┃  -30.0°        ┃  90.0°       ┃
┃ q5           ┃     0.07         ┃    0       ┃ 239.99999999999997°        ┃ -180.0°       ┃ 180.0°      ┃
┃ q6           ┃ -0.05831      ┃ 0.05     ┃                0.0°                       ┃  -60.0°        ┃  30.0°       ┃
┗━━━━━━━┻━━━━━━━━━━┻━━━━━━┻━━━━━━━━━━━━━━━━━━━━━┻━━━━━━━━━┻━━━━━━━━┛
│base │ t = 0, 0, 0; rpy/xyz = 0°, 90°, 0°     │
│tool │ t = 0, -0.005, 0; rpy/xyz = 0°, 0°, 0° │

Is it possible to include pre-rotated bases ? Here mine is pre-rotated of Ry(90°), and is it possible to include an extra tool location ?

Finally, I tried to run it without these extra settings and it return an error: 

>  [  1%] [................................................] Sum of Angles:  found new 'joint' (sumofangle) variable:  th_56
sum_of_angles_sub: created new equation: th_56 = th_4 + th_5
:  found new 'joint' (sumofangle) variable:  th_456
sum_of_angles_sub: created new equation: th_456 = th_3 + th_4 + th_5
Traceback (most recent call last):
  File "/home/puchaud/Projets_Python/IKBT-main/ikSolver.py", line 108, in <module>
    [M, R, unknowns] = kinematics_pickle(robot, dh, params, pvals, vv, unknowns, testing)
  File "/home/puchaud/Projets_Python/IKBT-main/ikbtbasics/ik_classes.py", line 100, in kinematics_pickle
    R.sum_of_angles_transform(unks)  # find sum of angles
  File "/home/puchaud/Projets_Python/IKBT-main/ikbtbasics/ik_classes.py", line 320, in sum_of_angles_transform
    rhs, newj, newe= sum_of_angles_sub(self, rhs, variables)
  File "/home/puchaud/Projets_Python/IKBT-main/ikbtbasics/ik_classes.py", line 387, in sum_of_angles_sub
    nil.append( str(get_variable_index(variables, v)) )
  File "/home/puchaud/Projets_Python/IKBT-main/ikbtbasics/ik_classes.py", line 442, in get_variable_index
    assert found, 'Error: trying to get index of an unknown joint variable' + str(symb)
AssertionError: Error: trying to get index of an unknown joint variable-th_3

Process finished with exit code 1

Any idea on what I have done wrong ? 

Regards,

blake5634, 2021-06-18:
Please check that you are using "modified" or Craig-style DH parameters (note change if subscript i, i-1 in each row).  See ik_robots.py for many examples.   Also use a symbolic value for each non-zero DH parameter (no reason numerical shouldn't work, but its less tested that way).    You can modify the last row to include your tool offset.   Also you can compensate for base rotation by pre-rotating the target prior to applying your IK solution.

blake5634, 2021-06-18:
I think your DH param issue is causing the mistaken computation that th4-6 are parallel axes ("created new equation: th_456...").   If you really do have these 3 parallel axes, then switch to the fixUR5regression branch which I think is needed for 3 parallel axes but has not yet been merged.

Ipuch, 2021-06-18:
Thank you for your answer,

Indeed, I was using non-modified DH parameters. I switched. I tried to replicate as much as possible the implementation in ik_robots.py. 

None of my axes are paralleles, in my robot.

I think i've downloaded the master branch. I'll try RepairUR5Regression. Is it the branch with the last version of the code ?

Ipuch, 2021-06-18:
Here is the last error, I get which looks the same I had with the wrong DH parameters.
[image]

Ipuch, 2021-06-18:
I tried with the RepairUR5Regression branch and it raised the same error concerning an unknown variable which is -th_6. You'll find my configuration below. Let me know if you have any clue. 
`    if (name == 'KINOVA'):
        # standardize on the order
        # "alpha N-1, a N-1, d N, theta N' for the DH table columns.
        #   This is KINOVA O110 arm
        variables = [unknown(th_1),
                     unknown(th_2), unknown(th_3),
                     unknown(th_4), unknown(th_5),
                     unknown(th_6)]

        params = [a_4,
                  l_3, l_4, l_5, l_6]
        pvals = {a_4: 0.25,
                 l_3: 0.1, l_4: 0.01, l_5: 0.07, l_6: -0.5831}

        dh = sp.Matrix([
            [-sp.pi / 2, 0, th_1, 0],
            [sp.pi / 2, 0, th_2, -sp.pi / 2],
            [sp.pi / 2, 0, l_3, th_3],
            [sp.pi / 6, 0, l_4, th_4],
            [0, a_4, l_5, th_5],
            [4 * sp.pi / 3, 0, l_6, th_6]
        ])
        # prismatic 0
        # revolute 1
        vv = [0, 0, 1, 1, 1, 1]
        `

blake5634, 2021-06-19:
OK, well you have two alphas for which cos(alpha_i)*sin(alpha_i) != 0  (i.e. non-multiples of pi/2)    This is of course generates dramatically more complex kinematic equations and that's basically untested territory for IKBT.     Suggested approach: try a very simple wrist-like robot (having two non N*pi/2 alphas and ideally with a known solution) and do some testing.

Ipuch, 2021-06-19:
Well, this is the structure of my arm. I didn't see any mention on the specification of having alpha multiple of pi/2 to solve IK. Could you enlighten me on this ? 

Do you think it would require a lot of adjustments in the code to solve such problems?

blake5634, 2021-06-20:
It's just that most (but not all) robot designs restrict the alphas to multiples of Pi/2.    If alpha is not nPi/2, then both sine and cosine terms are non zero.   This doubles the complexity of equations for each affected joint and then as the FK equations are computed the problem's scale kind of explodes.   Although we haven't formally studied it, it's not really a computing performance issue or a bug, it requires new solution methods. 
  Probably best approach for these types of 'bots is to use some of the many fine numerical packages that are out there.

Ipuch, 2021-06-21:
Thank you for your reply and advice. Hence, it would have been great to use this open access tool to solve my problem. I'll go for a try with a symbolic package such as Maple or symbolic matlab. Otherwise, I'll do numerical resolution.

Regards,
Pierre

blake5634, 2021-06-23:
If you get a good symbolic solution with Maple or Symb. Matlab, I'd love to see it for my edification.   Good luck.
BH

closed 2021-06-23 by blake5634