uw-biorobotics/IKBT #46: Panda Robot IK Solutions

aabouman, 2022-06-15:
Thanks for a really sweet package. I have been trying to compute the analytical solution for the first 6 joints of the Franka Panda robot with DH Parameters listed [here](https://frankaemika.github.io/docs/control_parameters.html). 
Adding the following to `ik_robots.py`:
```python
    if(name == 'Panda'):

        dh = sp.Matrix([                  ##  This one requires sum-of-angles.
        [  0,         0,  l_1,  th_1],
        [-sp.pi/2,    0,    0,  th_2],
        [ sp.pi/2,    0,  l_3,  th_3],
        [ sp.pi/2,  a_1,    0,  th_4],
        [-sp.pi/2,  a_2,  l_5,  th_5],
        [ sp.pi/2,    0,    0,  th_6]
        ])
        vv = [1,1,1,1,1,1]
        sp.var('a_1 a_2 l_1 l_3 l_5')
        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]

        params = [a_1, a_2, l_1, l_3, l_5]
        pvals = {a_1:0.0825, a_2:-0.0825, l_1:0.333, l_3:0.316, l_5:0.384}
```

and running `python3 ikSolver.py Panda` results in the error: 
```

          Running IK solution 

             Working on Panda

kinematics pickle: trying to open  fk_eqns/Panda_pickle.p  in  /home/alexander/work/SymbolicIK/IKBT
\Trying to read pre-computed forward kinematics from fk_eqns/Panda_pickle.p
Successfully read pre-computed forward kinematics
pickle contained  6  unknowns
GOT HERE: robot name:  Panda
Ticking IK BT for  Panda  -------------------------

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4

Assigner: variable on blackboard: th_5

Assigner: variable on blackboard: th_6

Assigner: variable on blackboard: th_1

Assigner: variable on blackboard: th_2

Assigner: variable on blackboard: th_3
x2y2 did not find suitable eqns

Assigner: variable on blackboard: th_4
[]
matching.py: bad notation collection
```

From a previous issue I figured this `matching.py` error ultimately means the solver is unable to compute a solution however I just wanted to check. Thanks!

blake5634, 2022-06-16:
Thanks for your interest.    I like the challenge of "unsolvable" robots
because most commercial designs have analytical solutions these days.   If
I get some time I'll take a look.
Blake

blake5634, 2026-08-24:
An update on this interesting robot.    I am working with Claude (it's a new era!) on enhancements.   When it's existing tree of solvers cannot solve the robot, it now exits with cleaner messages.  Rerunning Panda, this is a solution failure, not a bug.     Watch for new features relevant to this coming soon.