uw-biorobotics/IKBT #48: Kinematics with wrist displacement

mvoelk, 2023-01-24:
I am trying to find a solution for the Fanuc CRX series.

The kinematics of these robot is shown in the image below. The frames in the image are from an URDF file, not those resulting from the DH formalism.
[image]

Notable for this kinematics (like other, Kinova Jaco, GMF P150, Fanuc P250iB and Yaskawa Motoman HC10) is the wrist displacement to avoid the wrist singularity. This also makes the Fanuc CRX an cuspidal robot (https://arxiv.org/pdf/2210.05204.pdf, Sec. 5.4.2).

I've made several attempts to find a solution with IKBT, playing with the modified DH parameters where it is possible, but without success.

Is it in general possible for IKBT to find a solution for this kind of kinematics?

blake5634, 2023-01-31:
Markus,
   Thanks for reaching out.    Maybe post your DH parameters.   From the graphic it appears that you might have three intersecting axes (at the shoulder) but I'm not sure about the color code of the arrows.   If your robot has three intersecting axes then their is a solution according to the proof of Piper and Roth.   However I don't have any way to know which arms IKBT fails to solve which are in theory solvable (except to try them).    I assume you do not have the wrist offset in the IKBT DH version.    Even though it can fit into the DH formalism, it simplifies the problem for IKBT to leave it off.  You can add it to the solution later if it solves it.    Best of luck. 
BH

mvoelk, 2023-03-17:
Thanks for reply and the great work. I have to apologize for the late response...

The color coding of the axes is RGB corresponds to XYZ and the frames were directly taken form the URDF file. The axis of rotation is the fourth column in the table.

Since the kinematics have neither three intersecting nor three parallel axes, I gave up and went for a numerical solution. An analytical solution would of course be much better and may still relevant for others.

blake5634, 2023-03-19:
I have no knowledge of URDF but I hear its not straightforward to convert to modified DH parameters (do-able of course).  This is old knowledge however, maybe somebody has created an automatic translator.    At any rate, if you get back to IKBT double check your DH conversion.  Best of luck with your project.

closed 2023-03-25 by blake5634