uw-biorobotics/IKBT #49: Error getting index of unknown variable

Roozter, 2023-10-18:
Hi,

Im trying to get a custom robot's IK solution and I am running into the following error. 

`Starting Forward Kinematics
Completed Forward Kinematics
Starting Sum of Angles scan (slow!)
ik_classes: length Robot.mequation_list:  8
Starting sum-of-angles scan. Please be patient
 [  1%] [................................................] Sum of AnglesTraceback (most recent call last):
  File "ikSolver.py", line 111, in <module>
    [M, R, unknowns] = kinematics_pickle(robot, dh, params, pvals, vv, unknowns, testing)
  File "/home/smaghsoudi/IKBT/ikbtbasics/ik_classes.py", line 104, in kinematics_pickle
    R.sum_of_angles_transform(unks)  # find sum of angles
  File "/home/smaghsoudi/IKBT/ikbtbasics/ik_classes.py", line 350, in sum_of_angles_transform
    rhs, newj, newe= sum_of_angles_sub(self, rhs, variables)
  File "/home/smaghsoudi/IKBT/ikbtbasics/ik_classes.py", line 411, in sum_of_angles_sub
    nil.append( str(get_variable_index(variables, v)) )
  File "/home/smaghsoudi/IKBT/ikbtbasics/ik_classes.py", line 474, in get_variable_index
    assert found, 'Error: trying to get index of an unknown joint variable' + str(symb)
AssertionError: Error: trying to get index of an unknown joint variable-th_3
`
I'm not sure what the issue is, it doesnt take long before this error shows up after running the solver.

Here are my dh parameters.

`dh = sp.Matrix([                  
        [  0,          0,     d_1,     th_1  ],
        [-sp.pi/2,    a_1,     0,      th_2  ],
        [ sp.pi,      a_2,     0,      th_3  ],   
        [ sp.pi/2 ,   a_3,    d_2,     th_4  ],      
        [ sp.pi/2 ,    0,     d_3,     th_5  ],
        [-sp.pi/2 ,    0,     d_4,     th_6  ]
        ])
        vv = [1,1,1,1,1,1]

        variables =  [unknown(th_1), unknown(th_2), unknown(th_3), unknown(th_4), unknown(th_5), unknown(th_6)]

        params = [d_1, a_1, a_2, a_3, d_2, d_3, d_4]
        pvals = {d_1:0.112891 , a_1:0.12893 , a_2:0.8382, a_3:0.064 , d_2:0.4615 , d_3:0.029, d_4:0.418}`

Thanks!

blake5634, 2023-10-18:
This may have something to do with a corrupted pickle file.   Try removing the directory fk_eqns/ in your project and run again.

Roozter, 2023-10-18:
I tried removing but I'm getting the same error.

blake5634, 2023-10-19:
A couple of more thoughts.   First it is possible that no solution exists but of course that should probably not cause an assertion to fail.    If I understand your DH table right (it's been a couple years since I taught DH param assignments)  you do not have three consecutive intersecting axes (a sufficient condition for solubility).  However there is no known necessary condition so we can't be sure (or code a test for it!).  

Finally, although I'm pretty sure it is not the cause of your problem, your naming conventions for the d_i (joint offset parameter) might be problematic.   There may be code in IKBT somewhere that thinks e.g. d_4 is an unknown.  Since it seems to be not an unknown in your model, could you change it to ld_4 or something and change it in params?    Also for reference, the subscript of d_i should be same as the frame index (starting from 1) as you have with th_i.

blake5634, 2023-10-24:
I did a little more work on this.   I changed the var names as proposed above, and as a workaround, I changed \alpha_2 from sp.pi --> 0.   this should reverse that troublesome '-' on th_3 without fundamentally changing your robot.    It did and the search continued but ultimately gave the dreaded "bad notation collection" error.   Further work revealed that this error simply means no solution was found (which is a thing that can legit happen), so I edited some messages for a more graceful exit-on-failure-to-solve.

Roozter, 2023-11-07:
Thank you for looking at this professor. After pulling my hair out for a week or so, I have been unsuccessful in finding a solution by other means. Maybe there isnt a solution for this configuration after all. I'm going to close this issue.

closed 2023-11-07 by Roozter