uw-biorobotics/IKBT #51: cpp and python can not run

kelejichi6463, 2025-03-24:
Hi Prof. Hannaford,

The cpp code cannot run: 

g++ IK_equationsWrist.cpp 
IK_equationsWrist.cpp: In function ‘int ikin(double (*)[4], double (*)[6])’:
IK_equationsWrist.cpp:75:5: error: ‘Av1’ was not declared in this scope; did you mean ‘As1’?
   75 |     Av1 = atan2(r_21, r_11);
      |     ^~~
      |     As1
IK_equationsWrist.cpp:80:5: error: ‘Av2’ was not declared in this scope; did you mean ‘Bv2’?
   80 |     Av2 = atan2(r_21, r_11);
      |     ^~~
      |     Bv2

blake5634, 2025-03-25:
Hi,   The CPP code has always needed a little cleanup to run.    That's a simple syntax error.   Maybe ChatGPT or Claude can clean it up!  Not sure when I'll get around to fixing that.   Thanks for reporting it though. 
BH

kelejichi6463, 2025-03-25:
Professor, you are right. GPT help a lot. I would like to ask, in the case of KR16, there are joint angles for 2-6, but there is no calculation formula for th_1s1. Why is that?